OctoFarm/OctoFarm #1189: Wrong Times Displayed Throughout Web Interface

enz1ey, 2022-08-10:
### OctoFarm Version

1.6.5

### OctoPrint Version(s)

1.8.2

### pm2 or docker?

docker

### Operating System

Raspberry Pi OS

### Location of the problem

Web Interface

### Description of the problem

```markdown
So I have been seeing incorrect ETA times sprinkled throughout the web interface. I am running OctoFarm in a docker container, and I have three separate OctoPrint instances configured. The ETA in the "Current Operations" panel in the camera view is typically incorrect, and the times in the list view are also incorrect.
```

### Steps taken to reproduce

Check either of the camera or list views to compare ETA times.

### What did you expect to happen

I would expect the times to be reported accurately and consistently.

NotExpectedYet, 2022-08-10:
The times generated should all be to whatever your OS is set at for its timezone. What do you have those set at?

enz1ey, 2022-08-10:
> The times generated should all be to whatever your OS is set at for its timezone. What do you have those set at?

Time zones on both the rPi running the web client display as well as the Docker container running OctoFarm are set to America/New_York. I really don't think it's a local time issue, since you can see in the screenshot I linked, the times are different between the two panels on the same page. The "current operations" panel shows one ETA, the list panel shows a different time.

Also, the various panels (list/camera) show "no file selected" when the printers are printing. Sometimes a page refresh solves this, but I can't really do that easily since I'm using OctoFarm solely to display a page on a small screen on the wall near my printers which is being driven by a Pi with no keyboard/mouse connected.

NotExpectedYet, 2022-08-10:
I can't attached a picture over mobile for some reason but the screenshot you sent was blurry when I zoomed in to check the times on my phone. I'll check when I next get to my pc.

NotExpectedYet, 2022-08-10:
Something's not calculating right by the sounds though if it's the case. I'll check over that code and see what it does when I look, some of its not been touched in a while. 

Cheers for the report will get it sorted I'm sure.

enz1ey, 2022-08-10:
Yeah the picture uploaded smaller than expected for some reason. If you just add ".jpg" to the end of that link, it should render a larger image.

I am thinking maybe the OctoFarm Docker container might've been using UTC, because in that screenshot the hours were about three off exactly. But I know on my Pi display (can't grab a screenshot there unfortunately) they were off in the minutes field as well. I'm not home to check that display right now though.

What about the "no file selected" text in the list view? It also shows sporadically in the camera view, but the list view almost always shows "no file selected" even while the printers are printing...

NotExpectedYet, 2022-08-10:
Yeah something's not right with the dates. 

The no file selected one, would be something in the backend I gather. 

Unless there are any errors in the browser window but a log dump from the system page would help with that too see what's going on.

NotExpectedYet, 2022-08-10:
Those both use the same event stream as well so the information should be represented the same. 

Has this changed recently? Like maybe around the 1.4.x branches...

enz1ey, 2022-08-10:
I'm not seeing any errors in the browser console. I will grab a log dump, is anything sensitive contained in there that might need to be sanitized or anything? API keys, etc?

I also just started using OctoFarm the other day, so I've only been on this latest release.

enz1ey, 2022-08-10:
Okay so it seems perhaps the time zone wasn't correctly set in the OctoFarm Docker container after all, after specifying that in the compose file and recreating, the times are consistently represented where they should be.

HOWEVER, when starting a new print, the "Current Operations" panel ETA doesn't update. It looks like it just shows the print start time and never changes. Like I said, I have a Pi displaying the "list" page on a small display with no keyboard/mouse, so refreshing the page manually isn't an option. Below is what I'm seeing just after starting a new print on my Ender 6... The "Current Operations" panel shows (I think) the start time, while the "list" panel updates to show the correct ETA.
[image]

NotExpectedYet, 2022-08-11:
Thanks for the additional information. 

Ah yeah docker you need to explicitly set it but glad that's working at least. 

I gather your prints are started from OP if your just using it for display? 

I'll double check how that's handled because my usual testing flow doesn't account for that. 

I'll update the title for this and do another issue for the file name one. 

Would you mind uploading a log dump out of system page please? System -> Logs -> log dump. Will just help to make sure nothing else is going awry in the backend.

Didn't know if you knew but you can disable the pages you don't use as well in System -> Server -> Monitoring.

enz1ey, 2022-08-12:
Sorry for the delay. Yes, I start my prints from OctoPrint directly, OctoFarm is used solely for an informational display. Here are my logs:
[octofarm_dump-8-12-2022-9-11-29 AM.zip](https://github.com/OctoFarm/OctoFarm/files/9316494/octofarm_dump-8-12-2022-9-11-29.AM.zip)
.

enz1ey, 2022-10-20:
Just wanted to let you know @NotExpectedYet that this all seems to be working for the last couple weeks now! The screen updates as expected, information stays up-to-date, file names appear correctly. I no longer have to constantly reboot the Pi that drives my display to get the information to update when starting new prints, etc.

NotExpectedYet, 2022-10-20:
Awesome thanks for letting me know @enz1ey I found a right goof in the code that was causing your issue 😅. I was more confused how it was working for me afterwards.

commit 3a88d3ac (2022-09-04): fix(client): current operations wasn't updating finish time

Closes: #1189