OctoFarm/OctoFarm #1191: Reference Error

whjoyjr, 2022-08-29:
### OctoFarm Version

1.6.5

### OctoPrint Version(s)

1.8.2

### pm2 or docker?

pm2

### Operating System

Raspberry Pi OS

### Location of the problem

Not sure

### Description of the problem

```markdown
When Loading Dashboard the following Dialog Box opens:

ReferenceError ERROR (undefined): 
Please report this error to OctoFarm Issues!
Cannot access uninitialized variable.
FILE INFO
LINE: undefined
COL: undefined
STACK:
@http://farmpi.local/assets/dist/dashboard.runner.1.6.5.min.js:2:32150 asyncFunctionResume@[native code] @http://farmpi.local/assets/dist/dashboard.runner.1.6.5.min.js:2:35467 asyncFunctionResume@[native code] @http://farmpi.local/assets/dist/dashboard.runner.1.6.5.min.js:2:29392
```

### Steps taken to reproduce

1. Launch OctoFarm
2. When OctoFarm loads the dialog presents itself
3. dismiss and navigate to a different tab, no issues.
4. Return to Dashboard tab and dialog box returns
[image]

### What did you expect to happen

Dashboard to load normally

### To avoid the problem I had to

Dismiss the dialog box
[octofarm_dump-8-29-2022-5-25-42 PM.zip](https://github.com/OctoFarm/OctoFarm/files/9448135/octofarm_dump-8-29-2022-5-25-42.PM.zip)

NotExpectedYet, 2022-09-01:
Cheers for the report. I'll see if I can reproduce with my install. 

I think some of this might be related to the change in compiling the client but not sure. Seems a few have cropped up like this since the change.

whjoyjr, 2022-09-01:
In case it matters, this was with MacOS (latest) and Safari.  I can install Chrome and test if you like.  I do not have a Windows machine to test against.

NotExpectedYet, 2022-09-01:
No testing needed but thanks for the offer.

NotExpectedYet, 2022-09-04:
This is something specific to Safari I believe as I can't reproduce at all on my windows/linux builds. 

I've checked through the code and nothing is unsubstantiated as the error implies. I've updated some browser lists and hopefully that will allow the site to compile properly for what's needed by Apple. The issue will close when the release is built, but please do open the issue if it persists.

closed 2022-09-04 by NotExpectedYet