OctoFarm/OctoFarm #1192: Printer list page (/mon/list) not displaying accurate information

oldskewlcool, 2022-08-31:
### OctoFarm Version

1.6.5

### OctoPrint Version(s)

1.8.1, 1.7.2

### pm2 or docker?

pm2

### Operating System

Debian

### Location of the problem

Web Interface

### Description of the problem

```markdown
The printer list view (/mon/list) is basically unusable due to printer statuses not being accurately displayed across most columns.

Specifically the following things happening:

- Current File column displays mostly No File Selected
- Manage column icons are mostly greyed out
- Elapsed column mostly says "Loading"
- Progress mostly says "0%"
- Tools column mostly has no information to display

I say _mostly_ as intermittently a few things happen:
- One or two machines may populate the Status and Current File column though with stale inaccurate data
- The Elapsed column drifts in and out of accurate readings of elapsed time though not across all machines
- Manage column, if page is refreshed many times, will start to function and allow new files to be uploaded

As such you have no idea what machines are working on, their last files or the status of jobs etc..

The strange part is that the camera page (/mon/camera) has all of the information above across all machines accurately displayed.

This leads me think this is not a dupe of #1163 but is some sort of UI bug as accurate data does appear to be flowing into OctoPrint consistently - just not the printer list view.

Attached are screenshots and log dump.
```

### Steps taken to reproduce

Install OctoFarm 1.6.5
Connect many Prusa MK3S+ printers via OctoPi
Load /mon/list
All fields will mostly not populate or populate with incorrect information
Refreshing the page multiple times eventually enables the icons of the Manage column to function
Reconnecting all sockets, rebooting server, refreshing page all have no impact on the other fields

### What did you expect to happen

All columns to be populated with accurate data and all Manage buttons to be functional.

Data on the printer list page consistent with the camera view page.

[octofarm_dump-8-31-2022-9-33-30 AM.zip](https://github.com/OctoFarm/OctoFarm/files/9461938/octofarm_dump-8-31-2022-9-33-30.AM.zip)
[image]

### To avoid the problem I had to

No workarounds seem to be possible other than aggressive page refreshes to enable the manage buttons.

Steps taken to address / mitigate

- Try different OctoPrint versions
- Use different browsers (Firefox, Chrome)
- Log in / out
- Reboot server
- Reconnect all printer sockets
- Switch tabs across the web UI

NotExpectedYet, 2022-09-01:
Cheers for the report. That's a bit confusing as they all use the same data streams for each view.

Firefighter1023, 2022-09-01:
I confirm that I am having the same symptoms on a fresh installation of 1.6.5 via farmpi image on raspberry pi 4 hardware.  I have also noticed that the octofarm server is now randomly unable to write file or create folders to each octoprint instance.

NotExpectedYet, 2022-09-01:
@oldskewlcool are you on farm pi too?

Firefighter1023, 2022-09-01:
> @oldskewlcool are you on farm pi too?

Yes.  Previously I was on 1.2.6, I manually updated it to 1.6.5 by command line and found these symptoms.  I abandoned that build and did a fresh install using the nightly image from the farmpi GitHub, same symptoms arose.  I did not restore any databases with the fresh install.

ndarrow, 2022-09-01:
I see this problem too with FarmPi nightly install and OctoFarm version 1.6.5. I didn’t see it until I navigated off list view and back to list view. Clicking the “List” link button a few times does manage to get the values to populate but it doesn’t look like they automatically continue to update. You have to spam the List button to get updated data to show.

NotExpectedYet, 2022-09-02:
Weird how it's got to be spammed to get it to load properly. Sounding like something caching in the browser more than anything. 

Has anyone tried a different browser to see if it's continuing on that as well?

NotExpectedYet, 2022-09-02:
Don't think it's farm pi at least as I'm getting weird client reports of things that work fine for me locally and at my work instance. 

Will get it sorted

Firefighter1023, 2022-09-02:
> Weird how it's got to be spammed to get it to load properly. Sounding like something caching in the browser more than anything.
> 
> Has anyone tried a different browser to see if it's continuing on that as well?

Trying in Chrome as well as Explorer/Edge results in the same symptoms.  I can not get a full display load out with with spamming the list or refresh.  Only the top summarized display row consistently shows correct data.

NotExpectedYet, 2022-09-03:
> Trying in Chrome as well as Explorer/Edge results in the same symptoms. I can not get a full display load out with with spamming the list or refresh. Only the top summarized display row consistently [...]

Not that then cheers for trying.

NotExpectedYet, 2022-09-04:
This wont close when the next release is done shortly as I'm well unsure what's actually causing this. 

I've tested fresh installations of docker, pm2, farmpi and none exhibit this issue so I believe it's a combination of some other things I'm hoping are fixed. 

I've also tested updating from V1.2, V1.5 -> V1.6.5 and none would reproduce the issues. grrr! 

v1.6.6 should hopefully repair whatever goofed up with the client compilation though. 

Do let me know if this isn't resolved please.

Firefighter1023, 2022-09-04:
> This wont close when the next release is done shortly as I'm well unsure what's actually causing this.
>
> I've tested fresh installations of docker, pm2, farmpi and none exhibit this issue so I [...]

Mentioned in Discord, but brought here for logging purposes.  I manually updated from 1.6.5 to 1.6.6; the same symptoms remain in the updated server as did 1.6.5.

ndarrow, 2022-09-07:
I updated from V1.6.5 (FarmPi) to V1.6.6 and the List View now attempts to load what seems to be hundreds of rows (I have 52 printers actually connected). See attached pic (look at the scroll bar) and log dump.

Dashboard Current Status show the correct number of printers.
X- Panel View shows too many rows like List View.
X- Camera List View shows too many rows like List View.
Group view shows the correct number of printers. 
X- Super List View shows too many rows like List View.
Current Operations View shows correct number of printers.
The Printers window properly shows only the 52 printers connected. 

[image]
[octofarm_dump-9-7-2022-5-48-39 PM.zip](https://github.com/OctoFarm/OctoFarm/files/9508494/octofarm_dump-9-7-2022-5-48-39.PM.zip)

wicada, 2022-09-07:
I have this same issue on 1.6.6

NotExpectedYet, 2022-09-09:
If anyone gets chance would they mind sending me a video of the following please? Will help a lot. 

Open your browser and load OctoFarm. 

Open up the network tab in the console. Easiest way is to press F12, will work on Firefox and Chrome, then navigate to the network tab when the console pops up and do whatever actions are needed to cause the issue. 

If possible I'd like to see the full paths of the network calls the browser is making when it loads that page. 

I'm making some time this weekend to take another stab at fixing this but I'm going blind here not been able to reproduce locally or see anything weird in your logs. 

Big thanks for the dumps so far though got some other fixes because of those.

KKriz, 2022-09-11:
Same issue, but mine only shows ''no file selected'' , Time elapsed and progress does show.

NotExpectedYet, 2022-09-19:
Thank you for those who sent me a video. I realise now where the issue lies. I'm doing some work fixing the client build process and then I'll get this sorted out. If it's not today will certainly be next weekend.

closed 2022-09-24 by NotExpectedYet

NotExpectedYet, 2022-09-24:
:tada: This issue has been resolved in version 1.6.8 :tada:

The release is available on [GitHub release](https://github.com/OctoFarm/OctoFarm/releases/tag/v1.6.8)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

Firefighter1023, 2022-09-24:
Updated to 1.6.8, appears to have fixed all of the "ghost" instances in the printers list, the "no file selected" in the CurrentFile column still persists.  The top horizonal ribbon "Current Operations" shows all correct information, below that in the vertical columns is where the bug persists.

NotExpectedYet, 2022-09-25:
Could you please open another ticket please for whatever still ails you. The cause of this is not the cause of that.

commit 24828ffe (2022-09-20): fix(client): printer pages not correctly testing visibility of printer

printers wouldn't update due to missing visibility detection on client

Closes: #1192
commit 7e85d4a0 (2022-09-24): fix(client): printer pages not correctly testing visibility of printer

printers wouldn't update due to missing visibility detection on client

Closes: #1192