OctoFarm/OctoFarm #1194: Filament Manager Throwing Error

bretterer, 2022-09-02:
### OctoFarm Version

1.6.5

### OctoPrint Version(s)

1.8.2

### pm2 or docker?

pm2

### Operating System

Ubuntu

### OctoPrint Plugins

filament manager, bed visualizer, camera settings, exclude region, multicam, printtimegenius, printer dialogs, printer notifications, slicer thumbnails

### Location of the problem

Web Interface

### Description of the problem

```markdown
TypeError ERROR (undefined):
Please report this error to OctoFarm Issues!
Cannot read properties of undefined (reading 'length')
FILE INFO
LINE: undefined
COL: undefined
STACK:
TypeError: Cannot read properties of undefined (reading 'length') at HTMLButtonElement. (http://10.12.13.150:4000/assets/dist/filament-manager.runner.1.6.5.min.js:2:586392)
```

### Steps taken to reproduce

1. set up postgres database on octofarm server
2. set up filament manager in octoprint to use the postgresql server on octofarm.
3. in Octofarm, add a new spool

### What did you expect to happen

I would expect the spool to be created in the system

[image]

NotExpectedYet, 2022-09-02:
Thanks for the report, seems I've created a load of client issues with the new compilation. Will get them sorted.

NotExpectedYet, 2022-09-04:
This should be fixed for the next version.

bretterer, 2022-09-04:
Sadly, after updating to 1.6.6, I am seeing the same issues

NotExpectedYet, 2022-09-20:
Couldn't reproduce without for the next release v1.6.7. 

Closing because next release removed the plugin support check #1202 for more information

commit 97b0b1e7 (2022-09-04): fix(server): refactored filament manager sync

Closes: #1194
commit 2cec91ac (2022-09-04): fix(server): refactored filament manager sync

Closes: #1194
closed 2022-09-20 by NotExpectedYet