OctoFarm/OctoFarm #1198: V1.6.5 cannot change printers' names

ndarrow, 2022-09-07:
### OctoFarm Version

V1.6.5 (farmpi-0.2.0-2022-8-31-nightly.img)

### OctoPrint Version(s)

V1.8.2, V1.8.1, V1.7.3

### pm2 or docker?

pm2

### Operating System

Linux 5.4.0-1069-raspi (farmpi-0.2.0-2022-8-31-nightly.img)

### OctoPrint Plugins

NA

### Location of the problem

Web Interface

### Description of the problem

```markdown
I cannot change the name of the printers via:
Printers > Printer Manager > Bulk edit
or
Printers > Manage > Edit

Restarting OF via restart button in System did not work. 
Rebooting the Pi did not work.

Printers were initially added via bulk import file (printers.json), and the names imported correctly. I just can't edit them in the UI after they are imported.
```

### Steps taken to reproduce

Attempt to change printer name.
Name doesn't change.
I've restarted OF and rebooted pi with no luck. 

### What did you expect to happen

Printer name would change

### To avoid the problem I had to

No workaround attempted, but I assume I could delete all printers and manually edit the printers.json file, then bulk import the printers again.

NotExpectedYet, 2022-09-08:
This is going to be related to #1192 I bet

NotExpectedYet, 2022-09-20:
I can't reproduce this on the next version, I gather it was fixed in 1.6.6 but I'm not sure. 

I wouldn't upgrade as I have issues on that version with client draws but will get fix out sometime this week

NotExpectedYet, 2022-09-20:
This will close when the next version is released. Please let me know if it's not the case, I'll need log dumps if it isn't :) thanks!

closed 2022-09-24 by NotExpectedYet