OctoFarm/OctoFarm #1199: Unknown spool in history tab

KKriz, 2022-09-11:
### OctoFarm Version

1.6.6

### OctoPrint Version(s)

1.8.2

### pm2 or docker?

pm2

### Operating System

Unraid

### Location of the problem

Web Interface

### Description of the problem

```markdown
I am getting an unknown spool in the history tab. However, if I open it, it does show the correct spool in the overview? If I only click save changes without actually changing anything, then it does show the correct spool in the history tab.
```

### Steps taken to reproduce

Open History Tab
Click the print with the unknown spool.
Save changes without changing anything
Now the correct spool is being displayed in the history tab

### What did you expect to happen

The correct spool to display and not ''unknown spool''

NotExpectedYet, 2022-09-24:
Would you mind uploading a log dump please so I can see what's going on? 

I think it should be resolved for the next push just wanted to confirm

NotExpectedYet, 2022-09-24:
Ok found the source of the issue. Will be fixed for next release. 

I think, sadly, any already captured in this state may need repairing manually but I'll look into automating that shortly.

closed 2022-09-24 by NotExpectedYet

NotExpectedYet, 2022-09-24:
:tada: This issue has been resolved in version 1.6.8 :tada:

The release is available on [GitHub release](https://github.com/OctoFarm/OctoFarm/releases/tag/v1.6.8)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit b1c806f2 (2022-09-24): fix(server): spools were not attaching profiles properly

Closes #1199