OctoFarm/OctoFarm #1226: Not able to create folders - "Sorry your folder couldn't be saved"

NotExpectedYet, 2022-09-25:
### Discussed in https://github.com/OctoFarm/OctoFarm/discussions/1225

<div type='discussions-op-text'>

<sup>Originally posted by **rajavel1990** September 25, 2022</sup>
I have installed the latest version 1.6.8 on my windows machine and i am not able to create folders for each of the printers. I get "Sorry your folder couldn't be saved" error.
[image]
</div>

Firefighter1023, 2022-09-25:
Confirmed same on my server, was present in 1.6.5.  API and CORS is set correctly.

NotExpectedYet, 2022-10-01:
This is weird as I'm getting unauthorized on my local version when creating a folder. 

Seems like it's OP sending that back, as my requests haven't changed. Just trying to narrow it down now.

NotExpectedYet, 2022-10-01:
:tada: This issue has been resolved in version 1.6.9 :tada:

The release is available on [GitHub release](https://github.com/OctoFarm/OctoFarm/releases/tag/v1.6.9)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit 72a4853c (2022-10-01): fix(client): unable to create new folders on op client

header wasn't getting sent to OctoFarm when a multi-part form

Closes: #1226
commit 61d4653f (2022-10-01): fix(client): unable to create new folders on op client

header wasn't getting sent to OctoFarm when a multi-part form

Closes: #1226
closed 2022-10-01 by NotExpectedYet