OctoFarm/OctoFarm #1227: Under ListView, machines are randomly showing "No File Selected" under CurrentFile Column

Firefighter1023, 2022-09-25:
### OctoFarm Version

1.6.8

### OctoPrint Version(s)

1.8.2

### pm2 or docker?

pm2

### Operating System

Debian/FarmPi

### Description of the problem

When printers are working or in standby status the result is the same.  Viewing the "list" view results in each of the printers randomly displaying the correct current file, or more often "No File Selected" in the current file column.  

The top ribbon under current operations displays all correct information.

### Steps taken to reproduce

Install OctoFarm 1.6.8
Connect many mutiple printers via OctoPi
All current file fields will mostly not populate or populate randomly
Refreshing the page multiple times results in random displays of the same information
Reconnecting all sockets, rebooting server, refreshing page all have no impact on the other fields

[octofarm_dump-9-25-2022-10-40-48 AM.zip](https://github.com/OctoFarm/OctoFarm/files/9640996/octofarm_dump-9-25-2022-10-40-48.AM.zip)

swindmiller, 2022-09-26:
I noticed this too on 1.6.5, just updated to 1.6.8 and Current Operation showed I had a printer printing. List view showed Complete for the printer with no file loaded. Refreshed the page and it now showed Printing with a progress but still showed No File Selected. A few minutes later I returned to the tab (didn't close it) and it showed everything correct. Hopefully this help troubleshoot it. 
Scott

NotExpectedYet, 2022-09-26:
What browsers are you both using?

NotExpectedYet, 2022-09-26:
Also Is there any sorting applied on the page? Or filtering? 

Again I can't reproduce this with 30 printers been drawn on list view so I gather something is different here. I returned back to the old drawing method before 1.6.5 as well. The only thing that remains is checking if a printer is in view on the page before updating it. 

I tested Firefox and Firefox developer edition and chrome developer edition and they were working fine with the implementation on my Linux box.

NotExpectedYet, 2022-09-26:
Let me know anyway I'll try get to it again this week.

Firefighter1023, 2022-09-26:
> What browsers are you both using?

Tested in Chrome and IE on windows, Safari on IOS; I will check Firefox on Linux when I have a moment.

Firefighter1023, 2022-09-29:
Same result in firefox.

NotExpectedYet, 2022-09-29:
How strange. Cheers for trying those. I'll double check it's sounding like this is the in view detection so I will disable it over weekend until I can make sure it's working on more than just my browsers.

NotExpectedYet, 2022-10-01:
Huh, I disabled the inviewport check last update... what the hell is going on here then.

NotExpectedYet, 2022-10-01:
Think I've actually found the issue! woop

NotExpectedYet, 2022-10-01:
Wow that was annoying. 

If anyone wants a work around until I get this next version released, then disable "Current Operations" on the view it's on.

NotExpectedYet, 2022-10-01:
:tada: This issue has been resolved in version 1.6.9 :tada:

The release is available on [GitHub release](https://github.com/OctoFarm/OctoFarm/releases/tag/v1.6.9)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit 3a104303 (2022-10-01): fix(client): conflict between current operation id and main views

Current operations file element id was conflicting with the views file element id

Closes: #1227
commit 1ed29776 (2022-10-01): fix(client): conflict between current operation id and main views

Current operations file element id was conflicting with the views file element id

Closes: #1227
closed 2022-10-01 by NotExpectedYet