OctoFarm/OctoFarm #1228: Ran out of disk space.  After fixing, can only see web GUI on Ubuntu, but no other computer on the network

theamzngq, 2022-09-26:
### OctoFarm Version

1.6.8

### OctoPrint Version(s)

1.7.3, 1.8.1

### pm2 or docker?

pm2

### Operating System

Ubuntu

### Description of the problem

I ran out of disk space.  After expanding the partition and file system and making sure octofarm was running by opening it in Ubuntu itself, learned that no other machines on the LAN can see the web GUI.

### Steps taken to reproduce

1. visit 192.168.0.75:4000 on ubuntu, works fine
2. try to visit the same URL from any other device on the LAN and it times out, I can see nothing
[octofarm_dump-9-26-2022-12-56-44 AM.zip](https://github.com/OctoFarm/OctoFarm/files/9644195/octofarm_dump-9-26-2022-12-56-44.AM.zip)

NotExpectedYet, 2022-09-27:
This isn't an OctoFarm issue this is something with your firewall so I would check that. 

I'm going to close this because it's not an actual issue with OctoFarm. 

Still happy to help troubleshoot but from what you've said and the logs it's can't be much else.

closed 2022-09-27 by NotExpectedYet

theamzngq, 2022-09-27:
The only strange thing is that it was working normally until Ubuntu ran out
of disk space.  At that moment, I could no longer access it on the LAN.
There was no firewall issue before that and I didn't touch anything else in
Ubuntu.  I will check to see if the firewall somehow turned itself on?  Or
reconfigured itself?

NotExpectedYet, 2022-09-27:
Have you set it up at all before then or just default? 

Yeah it is a bit of a weird one like, all I can think is during the space filling up either a permission goofed or a file didn't get written when it was supposed to. 

You can check the normal one ufw with `sudo ufw status` 

https://kerneltalks.com/virtualization/how-to-reset-iptables-to-default-settings/amp/

Might be worth running the commands in that article too and making sure there's nothing weird in iptables.

theamzngq, 2022-09-27:
I just disabled IPv6 and it started working again...strange, but I guess
that means I'm back in business!

NotExpectedYet, 2022-09-27:
Are there any other services running on the box? And do they have similar symptoms. 

I gather the above IP was the local IP that OctoFarm is installed on? 

I forget the routing but I "think" that should be blocked as well and only localhost should work. 

I checked the logs through though and there's no Mongodb errors, and I can see one of your successful connection attempts on the 26th I think it was.

NotExpectedYet, 2022-09-27:
Epic thanks for the update. 

Glad it's working again.