OctoFarm/OctoFarm #1229: Will not connect to any of my printers

bnieves1994, 2022-09-26:
### OctoFarm Version

latest

### OctoPrint Version(s)

1.8.3

### pm2 or docker?

pm2

### Operating System

Windows

### Description of the problem

I just set up octofarm but I have yet to connect a single printer. I am trying with my ender 3 S1 and my KP3S and neither is working it shows the api as green but then says "failed to acquire OctoPrint's printer connection data" 

Any help would be greatly appreciated and can gladly provide any logs and info needed please just ask. 

I see you all like log dumps so I am including everything

[octofarm_dump-9-26-2022-5-16-16 PM.zip](https://github.com/OctoFarm/OctoFarm/files/9649736/octofarm_dump-9-26-2022-5-16-16.PM.zip)

### Steps taken to reproduce

1. open octofarm
2. try to connect a printer
3. utter failure

NotExpectedYet, 2022-09-27:
It looks like you've got some network issues because every call my system is making is timing out at 10 seconds. That's a really long time to not get a response. 

I see you've already noticed the octopi.local addresses don't work very well aha. still not entirely sure why that is but IP will work fine. 

I'm not entirely sure what to suggest. 

Are these on OctoPi? Or have you installed your own instances? If it's OctoPi it should be without the port as it actively blocks the internal 5000 port. 

Yours are 5001 though which would have to be a custom OP setup which is why I ask.

bnieves1994, 2022-09-27:
I am running a multi-instance octopi. I have 2 instances currently deployed that I am trying to connect. I was able to get 5001 connected last night but am still facing the same issues with 5002. Both were failing for the same reason they are not able to access /api/users and /api/settings I can access both directly from the URL. Again 5001 is now working but same issue with 5002. I bumped my timeouts all the way up to try to get a connection. 

Please let me know any other info that may be helpful.

And thank you so much for your help and time!

bnieves1994, 2022-09-27:
Printers are an Ender 3 S1, connected, and Kingroon KP3S, not connected.

NotExpectedYet, 2022-09-27:
Can you please send an updated log dump through so I can see those attempts?

NotExpectedYet, 2022-09-27:
No trouble as well, I use a custom setup myself with a script so there's no reason this shouldn't be working.

bnieves1994, 2022-09-28:
Here is my newest log dump. 

I have both admin and owner permissions on my account I am using the user key from within OctoPrint if that is what you are asking. 

In connection manager it shows connection on everything but /api/users and api/settings

Lastly I am using octoprint deploy I know you said you are also running a custom instance setup but incase that helps at all I wanted to share it. The Pi has been restarted a few times today as well as both instances and :5002 is still connected and going strong. 

[octofarm_dump-9-27-2022-9-43-18 PM.zip](https://github.com/OctoFarm/OctoFarm/files/9660716/octofarm_dump-9-27-2022-9-43-18.PM.zip)

NotExpectedYet, 2022-09-28:
Thank you for the additional log dump, I'll take a look when I get a mo. 

Really weird the other isn't working but that is. I'm wondering if there's a bug somewhere as another user has a weird issue with his profiles not grabbing and sticking but on a single printer. 

Octo

NotExpectedYet, 2022-09-28:
Sorry dog jumped on me and I posted aha. 

Octo-deploy should be fine. 

Out of interest have you tried deleting the one that won't connect and re-adding it? Just a guess like until I've read the logs.

NotExpectedYet, 2022-10-01:
> Here is my newest log dump.
>
> I have both admin and owner permissions on my account I am using the user key from within OctoPrint if that is what you are asking.
>
> In connection manager it shows [...]

Just curious here, because OctoDeploy uses the haproxy configuration so port 5001 shouldn't work. It requires the use of the /print_name that it sets up for you unless you've disabled the firewall? 

Currently the /printer_name url method isn't supported so I'm fixing this today. 

All I'm seeing in the logs are 403 (forbidden) that's on port 5002, and 5001 is timing out a lot. Then for some reason one of your instances stopped receiving a pong from the client forcing it to reconnect quite a lot, every 10 seconds.

NotExpectedYet, 2022-10-01:
Relevant issue #1236 as I didn't get chance to work in this getting fixed.

bnieves1994, 2022-10-02:
Sorry for the delayed response was dealing with that storm. 

5002 is still connected and still nothing from 5001. I havent tried a new instance yet since the storm I have been trying to play catch up and the printers havent stopped. Some time tomorrow I can try to make a new instance for it on another port and report back. If you want me to try the /printer name method etc just let me know.

NotExpectedYet, 2022-10-02:
> Sorry for the delayed response was dealing with that storm.
>
> 5002 is still connected and still nothing from 5001. I havent tried a new instance yet since the storm I have been trying to play catch [...]

No worries, all ok I hope? 

No rush as well :) 

You shouldn't have to spin up a new instance. I think 5002 isn't working because of octo-deploy and firewall rules. I'm not privy with that system so you'd have to check yourself what it actually sets up. 

I know from another user the ports didn't work for them. 

There's a limitation in OctoFarm where whatever is on the end after the last / gets deleted when it gets converted to a websocket URL. The issue I linked above is for tracking that. So you can't try the /name URL yet.

NotExpectedYet, 2022-10-02:
Just to let you know I'm closing this as it's not really to do with OctoFarm apart from #1236 for tracking the actual issue. 

I'm happy to continue troubleshooting with you regarding the issue though.

closed 2022-10-02 by NotExpectedYet

bnieves1994, 2022-10-07:
I appreciate it man. I on a whim switched the farm to Klipper but the multi instance was too much for the pis to handle with klipper so I had to leave octoprint for now. But I seriously appreciate all your help!

NotExpectedYet, 2022-10-07:
Your most welcome. What hardware were you running it on? I had 4 klipper+OP running on an old dual core with a 4gb stick of ram fine. Pi4 seemed to handle two with that setup, never tried anymore.

bnieves1994, 2022-10-07:
I am running a pi3b I think thats why. I had constant stopping then starting during prints. I want to spin up a distro of linux and give it all a shot again with that, I have enough computers sitting around lol. I love the PI I just think for an instance like mine where I would ideally run 4 to 5 printers per pi a computer just makes more sense. But the issues were there with one marlin and one klipper printer on octoprint.

NotExpectedYet, 2022-10-07:
Hmm one instance shouldn't really do that to a Pi4 but yeah I agree. 

If you throw a little extra horse power you can shove OctoFarm on there too. Its not a supported configuration but I've run it myself at work quite successfully. That's the core duo, and the old SSD failed before the CPU or memory aha.

NotExpectedYet, 2022-10-07:
Ah wait the stopping and starting during prints. Like a pause because it's thinking about something? 

That's either the usb not been shielded or your marlin firmware had a low usb buffer set. 

Klipper should have helped with that, if it doesn't it has an option for a virtual SD card which should certainly help.