OctoFarm/OctoFarm #1233: Multi-Print Missing File Directories

cuesir, 2022-09-29:
### OctoFarm Version

1.6.8

### OctoPrint Version(s)

1.8.3

### pm2 or docker?

pm2

### Operating System

Windows

### Description of the problem

Multi-Print Missing File Directories even though all my octoprint instance have synchronized Gcode directories.

### Steps taken to reproduce

1. Rebooted octofarm.
2. Looked at file manager, files and directory is there.
3. Folder Missing only in multi-print directory search.

NotExpectedYet, 2022-10-01:
:tada: This issue has been resolved in version 1.6.9 :tada:

The release is available on [GitHub release](https://github.com/OctoFarm/OctoFarm/releases/tag/v1.6.9)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

NotExpectedYet, 2022-10-03:
Still missing some structures

commit 4a4465c7 (2022-10-01): fix(server): normalise filenames before comparing for uniqeness

Closes: #1233
commit e05c984b (2022-10-01): fix(server): normalise filenames before comparing for uniqeness

Closes: #1233