OctoFarm/OctoFarm #1234: Unable to update Wallpaper

cuesir, 2022-09-29:
### OctoFarm Version

1.6.8

### OctoPrint Version(s)

1.8.3

### pm2 or docker?

pm2

### Operating System

Windows 10

### Description of the problem

Unable to change wallpaper

### Steps taken to reproduce

1. Selected .jpg
2. Clicked upload
3. page refreshes and nothing happens.
4. restarted octofarm and wallpaper still not updated.
5. replaced the default_bg.jpg file to the file I want and still not changed.

NotExpectedYet, 2022-09-29:
I believe your old wallpaper should come back when I've fixed this. 

I gather history view is broken too

NotExpectedYet, 2022-10-01:
:tada: This issue has been resolved in version 1.6.9 :tada:

The release is available on [GitHub release](https://github.com/OctoFarm/OctoFarm/releases/tag/v1.6.9)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit a3091e5b (2022-10-01): fix(client): incorrect path for background image

Closes: #1234
closed 2022-10-01 by NotExpectedYet