OctoFarm/OctoFarm #1293: Printer Sort Index - Loading Wheel?

SuperTrollMan, 2022-10-04:
### OctoFarm Version

1.7.2

### OctoPrint Version(s)

1.8.4

### pm2 or docker?

pm2

### Operating System

Ubuntu 20.04.2 LTS

### Description of the problem

The first printer in the list shows a loading wheel instead of the sort index number.
[image]

If you drag printer 2 to the first location, both 1 and 2 slots read 1.
[image]

Then when you force refresh the browser, printer 2 is now in slot 1 as it should, but the wheel is still against the printer in that first slot.
[image]

### Steps taken to reproduce

1. Upgraded from 1.6.6 to 1.7.2
2. Noticed issue

NotExpectedYet, 2022-10-04:
huh weird cheers for the report!

NotExpectedYet, 2022-10-04:
Reproduced, lovely how I've not touched them files. Bet it's from a dependency update. 

It seems the backend is coping with it, and it's purely UI