OctoFarm/OctoFarm #1308: Error message pops up when trying to edit a printer that is failing to connect

JesNoFear, 2022-10-14:
### OctoFarm Version

1.7.3

### OctoPrint Version(s)

1.8.4, 1.8.4, 1.8.4

### pm2 or docker?

pm2

### Operating System

Debian 10 (because 11 wouldnt load the DB)

### Description of the problem

FILE INFO
LINE: 2
COL: 665150
FILE:
http://10.255.10.1/assets/js/printer-management.runner.1.7.3.min.js
STACK:
g@http://10.255.10.1/assets/js/printer-management.runner.1.7.3.min.js:2:665150 updateStateElements@http://10.255.10.1/assets/js/printer-management.runner.1.7.3.min.js:20:48641 Rt@http://10.255.10.1/assets/js/printer-management.runner.1.7.3.min.js:20:3298 @http://10.255.10.1/assets/js/printer-management.runner.1.7.3.min.js:20:255800 v/f.onmessage@http://10.255.10.1/assets/js/printer-management.runner.1.7.3.min.js:2:665692 

### Steps taken to reproduce

Login as admin
Click on Printers in the top right
Click Settings on printer #3
Error message pops up

JesNoFear, 2022-10-14:
[octofarm_dump-10-14-2022-12-25-41 AM.zip](https://github.com/OctoFarm/OctoFarm/files/9782537/octofarm_dump-10-14-2022-12-25-41.AM.zip)
[image]

NotExpectedYet, 2022-10-14:
Huh good catch. Thanks for the report.