OctoFarm/OctoFarm #1317: ERROR MESSAGE: Task 'INITIALISE_PRINTERS_TASK' threw an exception:TypeError: Cannot read property 'toString' of undefined

ldiegos, 2022-10-15:
### OctoFarm Version

1.7.3

### OctoPrint Version(s)

1.8.2; 1.7.2; 1.6.1

### pm2 or docker?

pm2

### Operating System

Windows

### Description of the problem

Hi,
Octofarm 1.7.3 is death, nothing happen, just have the web page and nothing else is going on.
Just updated from branch 1.1.2 

PM2.out.log
------------------------------------------------------
Task 'INITIALISE_PRINTERS_TASK' threw an exception:TypeError: Cannot read property 'toString' of undefined
2022-10-15T12:47:43:     at D:\Octofarm\server\services\filament-cleaner.service.js:195:53
2022-10-15T12:47:43:     at baseFindIndex (D:\Octofarm\server\node_modules\lodash\lodash.js:816:11)
2022-10-15T12:47:43:     at Function.findIndex (D:\Octofarm\server\node_modules\lodash\lodash.js:7322:14)
2022-10-15T12:47:43:     at Function.createStatistics (D:\Octofarm\server\services\filament-cleaner.service.js:194:25)
2022-10-15T12:47:43:     at Function.start (D:\Octofarm\server\services\filament-cleaner.service.js:82:46)
2022-10-15T12:47:43:     at processTicksAndRejections (internal/process/task_queues.js:93:5)
2022-10-15T12:47:43:     at async INITIALISE_PRINTERS_TASK (D:\Octofarm\server\tasks.js:50:5)
2022-10-15T12:47:43:     at async Function.timeTask (D:\Octofarm\server\services\task-manager.service.js:118:5)

Task 'FILAMENT_CLEAN_TASK' threw an exception:TypeError: Cannot read property 'toString' of undefined
2022-10-15T12:47:43:  [0;31mERROR MESSAGE: [1;37mTask 'FILAMENT_CLEAN_TASK' threw an exception:TypeError: Cannot read property 'toString' of undefined
2022-10-15T12:47:43:     at D:\Octofarm\server\services\filament-cleaner.service.js:195:53
2022-10-15T12:47:43:     at baseFindIndex (D:\Octofarm\server\node_modules\lodash\lodash.js:816:11)
2022-10-15T12:47:43:     at Function.findIndex (D:\Octofarm\server\node_modules\lodash\lodash.js:7322:14)
2022-10-15T12:47:43:     at Function.createStatistics (D:\Octofarm\server\services\filament-cleaner.service.js:194:25)
2022-10-15T12:47:43:     at Function.start (D:\Octofarm\server\services\filament-cleaner.service.js:82:46)
2022-10-15T12:47:43:     at processTicksAndRejections (internal/process/task_queues.js:93:5)
2022-10-15T12:47:43:     at async FILAMENT_CLEAN_TASK (D:\Octofarm\server\tasks.js:153:3)
2022-10-15T12:47:43:     at async Function.timeTask (D:\Octofarm\server\services\task-manager.service.js:118:5)
2022-10-15T12:47:43:     at async AsyncTask.asyncHandler [as handler] (D:\Octofarm\server\services\task-manager.service.js:109:7) 

[octofarm_dump-15-10-2022-13-01-40.zip](https://github.com/OctoFarm/OctoFarm/files/9791852/octofarm_dump-15-10-2022-13-01-40.zip)

### Steps taken to reproduce

Just look into the logs files after realizing that version 1.7.3 isn't doing nothing

ldiegos, 2022-10-15:
News about this:
After clicking on "force reconnect" for each printer...
[image]

It seems to start working.

NotExpectedYet, 2022-10-15:
From branch 1.1.2? 

If that's the case what is the nodejs and npm version your running? 

`node -v`
`npm -v`

Both should be at least node: 14.20.1, npm: 6.14.17

I apologize for that update, it got pushed before it was ready and I've not had chance all week to sort anything with the releasing.

ldiegos, 2022-10-15:
node -v => v14.16.1
npm -v => 7.11.2

The first error continue to be logged on the log file... and this is not about this errors, but the "filament menu" raise an error... also and can't configure the remote access to the postgress SQL database with all the filaments... 
[image]

The setting menú with the filament manager configuration disappeared..

NotExpectedYet, 2022-10-15:
https://github.com/OctoFarm/OctoFarm/issues/1202

Filament Manager plugin support was dropped in 1.7.0. I posted notices on release notes/github issues/discord and patreon about it. 

It seems upgrading from an instance with that enabled is causing issue. As I've had another report from a user with it too but he could access that page fine from his reports. 

The filament manager page isn't loading because of whatever is failing with the start up tasks to generate the statistics.

ldiegos, 2022-10-15:
Thanks so much. Is there any other option to have a unique place to see the spools and the total used filaments?

lol I have another error:
[image]

NotExpectedYet, 2022-10-15:
> Thanks so much. Is there any other option to have a unique place to see the spools and the total used filaments?
> 
> lol I have another error: [image]

I left the old interface in so you can still manage spools and track their usage. There's two options in System -> Server -> Filament Manager that allow you to downdate the spools when History is logged. The only difference is it now no longer bothers to update the plugins database. 

If these issues were not there anyway... 

I think these are going to be related to your filament issue, it's failing to parse them and all views try to load them... 

I can only gather there's a spool on your system that doesn't have a matching profile. 

Are you able to export your filament database with this menu option? Be good to take a look at it. 

[image]

I'm unsure of a workaround currently to get the interface loading again. Nuking your filament and restarting OF should solve it but I've no quick way to import them all back into the interface. If you can't wait for a fix then would be the best option currently.

NotExpectedYet, 2022-10-15:
[image]

Those are the Filament Manager options. The length OP reports isn't too far off what Filament Manager Plugin tracks anyway, depending on slicer it's just as good most of the time. 

This should be disabled by default but to replicate the old functionality with Filament Manager Plugin then make sure "Allow Multiple Select" is disabled. 

[image]

ldiegos, 2022-10-15:
Hi,
Here you are, the filament DB.
[FilamentDB.zip](https://github.com/OctoFarm/OctoFarm/files/9792355/FilamentDB.zip)
I can nuke everything, as long as it do not delete my external postgres DB... 

I'm going to shutdown my databaseVM and nuke the octofarm, just to be sure that i don't get a surprise lol

ldiegos, 2022-10-15:
Yeah... you are completly correct... nuking the filament... everything seems to work again.

NotExpectedYet, 2022-10-15:
I will investigate why that happened so other users don't get the same issue. Thanks for reporting back. Your database will help resolve the issue. 

Yeah it doesn't touch the postgres database so you can still manage that there if you want.

ldiegos, 2022-10-15:
Thank you so much @NotExpectedYet lets see if in a future @oligist could finish the SpoolsManager with an external database and you will be able to communicate with it.

NotExpectedYet, 2022-10-16:
> Thank you so much @NotExpectedYet lets see if in a future @oligist could finish the SpoolsManager with an external database and you will be able to communicate with it.

Your welcome. 

I think they're both dead tbh. I'm having to make a lot of concessions with plugin support as they can be high in technical debt and with me been the only developer on this project I need to focus on stability and more print management feature sets. 

I've no problem with a PR to implement these features.

ldiegos, 2022-10-17:
> I think they're both dead tbh. 

Indeed, I'm afraid so...