OctoFarm/OctoFarm #1323: Server side sorting and filtering of printers on a per user basis

NotExpectedYet, 2022-10-19:
### Discussed in https://github.com/OctoFarm/OctoFarm/discussions/1321

<div type='discussions-op-text'>

<sup>Originally posted by **Gonchitec** October 18, 2022</sup>
I would like to propose a new functionality, I don't know if it's a bug or if it hadn't been thought of before.
First of all I love Octofarm and I think you have done a great job so far, I have been able to see that if I try to launch machines from two different clients with two different users at the same time, the list of printers is updated on both computers, so if I filter by group "1" and I am launching the printers of group "1" and someone tries to launch printers at the same time as me of group "2" filtering in the list, the list is updated for both at the same time and the two clients change to category "2", therefore the filter changes for both computers, would there be any way that the filter of the list section was unique per user?
Thanks for everything and greetings</div>

Gonchitec, 2022-10-19:
ADD: I had this error when i try to sort, i dont know if this problem is for the server side sorting if i have 2 computers sorting at the same time but i let you know too

TypeError ERROR (undefined):
Please report this error to [OctoFarm Issues](https://github.com/octofarm/octofarm/issues)!
Cannot read properties of null (reading 'id')
FILE INFO
LINE: undefined
COL: undefined
STACK:
TypeError: Cannot read properties of null (reading 'id') at Dt (http://192.168.21.137:4000/assets/js/list-view.runner.1.7.3.min.js:2:703678) at http://192.168.21.137:4000/assets/js/list-view.runner.1.7.3.min.js:2:861764 at async Pi (http://192.168.21.137:4000/assets/js/list-view.runner.1.7.3.min.js:2:859195) at async http://192.168.21.137:4000/assets/js/list-view.runner.1.7.3.min.js:2:862907

Thx for all!

NotExpectedYet, 2022-10-19:
Hmm possibly. I'll look out for that thanks!