OctoFarm/OctoFarm #1326: Print completion by day: Cancelled jobs are assigned to previous day

ldiegos, 2022-10-20:
### OctoFarm Version

1.7.3

### OctoPrint Version(s)

Not important for this.

### pm2 or docker?

pm2

### Operating System

Windows

### Description of the problem

Checking the history of printings,
I count 7 cancelled printings. 
[image]

But when i check the graph, the cancelled are in a previous day.
[image]

Also related with the "Print completion by day", this graph shows most of the time incorrect values for each day. So i need to go to history, show all the printings, and then go back to the dashboard and the graph then update with the correct values... this happened almost all the time.

Logs:
[octofarm_dump-20-10-2022-22-25-59.zip](https://github.com/OctoFarm/OctoFarm/files/9833681/octofarm_dump-20-10-2022-22-25-59.zip)

### Steps taken to reproduce

Open octofarm web page.
Check the graph

NotExpectedYet, 2022-10-21:
Ooo good catch on that one will take a peak thanks!

ldiegos, 2022-10-21:
Thanks to you!

NotExpectedYet, 2022-10-22:
The graph shows or should show what's on the current pages you have loaded in history which isn't ideal. 

If you've any feature requests / ideas for history it's worth putting them in discussions. My patreons asked for history and filament as a focus for these next few months of development so it's a good time to get them in my head.