OctoFarm/OctoFarm #1329: Fresh install failing to boot server

SavvyRainbow, 2022-11-05:
### OctoFarm Version

1.7.3

### OctoPrint Version(s)

N/A

### pm2 or docker?

docker

### Operating System

Ubuntu

### Description of the problem

I just set up a new docker server and added octofarm to it. I followed the guide for setting up the sever with docker compose. The error I am getting is:
"The OCTOFARM_PORT setting was not a correct port number: >= 0 and < 65536. Actual value: undefined." 

Attached is my log dump from docker and my docker-compose.yaml file.

[_octofarm_logs.txt](https://github.com/OctoFarm/OctoFarm/files/9943110/_octofarm_logs.txt)
[docker-compose.yaml.txt](https://github.com/OctoFarm/OctoFarm/files/9943114/docker-compose.yaml.txt)

I am very new to docker so forgive me if I missed something obvious.

### Steps taken to reproduce

added config from the website to docker-config.yaml and received the error immediately after

NotExpectedYet, 2022-11-05:
Morning,

Yeah that's a lacking in my documentation. Internally the docker container uses the octofarm:octofarm user. 

You need to `sudo chmod 777 OctoFarm/` directory and it should boot with a restart.

SavvyRainbow, 2022-11-05:
Appreciate the response. that does seem to have fixed that problem but.... now there is another issue. attached is my current log. looks like it might be an issue with mongodb. I included my mongodb log just incase.
[_mongodb_logs (1).txt](https://github.com/OctoFarm/OctoFarm/files/9944217/_mongodb_logs.1.txt)
[_octofarm_logs.txt](https://github.com/OctoFarm/OctoFarm/files/9944219/_octofarm_logs.txt)

NotExpectedYet, 2022-11-05:
Weird, its saying the username and password is wrong but your example looks correct

SavvyRainbow, 2022-11-05:
I did nothing... but now it is working...

NotExpectedYet, 2022-11-05:
Love it, glad it's working!

closed 2022-11-05 by NotExpectedYet