OctoFarm/OctoFarm #1333: OctoFarm seems to require domain names (not just host names)

Daghis, 2022-11-16:
### OctoFarm Version

1.7.3

### OctoPrint Version(s)

1.8.6

### pm2 or docker?

docker

### Operating System

Ubuntu

### Description of the problem

I'm setting up an Ubuntu Docker host to run OctoFarm _and_ several OctoPrint instances. Because they are all connected in the same bridged Docker-internal network (`octonet`), I'd like to be able to access them by their simple container names.

For example, I have a printer named Tiger. I've set up a container of OctoPrint with that name. I should be able to access it from `octonet` as `http://tiger/`, however I get the rejection below:

```
OctoFarm Error ERROR (400):
Please report this error to [OctoFarm Issues](https://github.com/octofarm/octofarm/issues)!
Bad request:
###ValidationException###

#PRINTERURL#
The printer url format is invalid.
FILE INFO
LINE: undefined
COL: undefined
STACK:
OctoFarm Error: Bad request:
###ValidationException###

#PRINTERURL#
The printer url format is invalid.
at http://farmhouse.local/assets/js/printer-management.runner.1.7.3.min.js:2:667626 at async Kn.savePrinter (http://farmhouse.local/assets/js/printer-management.runner.1.7.3.min.js:20:213822) at async HTMLButtonElement. (http://farmhouse.local/assets/js/printer-management.runner.1.7.3.min.js:20:211415)
```

When I replace `tiger` with its IP address on `octonet`, everything works well. Is using a bare hostname the problem here?

### Steps taken to reproduce

0. Install Ubuntu and docker-ce.
1. Set up Mongo and OctoFarm in Docker.
2. Set up an OctoPrint instance in Docker on the same bridged network using a single word name for the container (and resulting host name on the network).
3. Attempt to set up OctoFarm to connect to `http://_hostname_/`.
4. Observe error states.

Daghis, 2022-11-16:
As a workaround, if I add the printer using its `octonet` IP address, I can then later edit it to replace the IP address with its short-form name. Things do continue to work after that.

Daghis, 2022-11-16:
I believe this is the error inside the system log:

```
11/16/22, 1:44:17 PM | INFO | Middleware-Validation 
 INFO MESSAGE: 
      Color is Valid: true,
      Colour Transparent is Valid: true,
      Default Language is Valid: true,
      Name is Valid: true,
      Show Fahrenheit Is Valid: true
     
11/16/22, 1:44:17 PM | ERROR | Middleware-Validation 
 ERROR MESSAGE: Invalid body input detected in http://farmhouse.local/printers/add : {"name":"ValidationException","errors":{"printerURL":{"message":"The printer url format is invalid.","rule":"url"},"camURL":{"message":"The cam url format is invalid.","rule":"url"}}} 
11/16/22, 1:44:17 PM | HTTP | Middleware-Morgan 
 HTTP MESSAGE: POST /printers/add 400 183 - 5.478 ms
```

NotExpectedYet, 2022-11-18:
Hi thanks for the report. 

Currently OctoFarms host system needs to know how to route whatever your telling OctoFarm about, that should be all but I often have to tell users to use IP addresses or proper DNS not mDNS as nodejs doesn't seem to like it. 

The errors are just the validation in the API endpoint. I gather it's expecting a TLD in whatever package I'm using so yeah looks like I should just have to relax or tweak that check to fix this. For now a .local should get around it.

NotExpectedYet, 2022-11-18:
> As a workaround, if I add the printer using its `octonet` IP address, I can then later edit it to replace the IP address with its short-form name. Things do continue to work after that.

Haha inconsistent validation, nice spot I'll sort that too.

Daghis, 2022-11-18:
I don't think .local is not an option as that resolves via mDNS whereas these are just host names managed in /etc/hosts via Docker, no TLD required.

Perhaps if I named the containers with .local that might work, but that'd be a workaround/hack. Requiring a period restricts basic hostname-level access. That validation may help most, but it does present a small issue in this particular setup.

All of this said. I'm putting this together myself from scratch. There may be guides or "best practices" that suggests a better farm setup with several printers connected to a single PC.

NotExpectedYet, 2022-11-18:
> I don't think .local is an option as that resolves via mDNS whereas these are just host names managed in /etc/hosts via Docker, no TLD required. I don't know whether the containers support mDNS resolution or are broadcasting their own existence via mDNS.
> 
> Perhaps if I named the containers with .local that might work, but that'd be a workaround/hack. Requiring a period restricts basic hostname-level access. That validation may help most, but it does present a small issue in this particular setup.
> 
> All of this said. I'm putting this together myself from scratch. There may be guides or "best practices" that suggests a better farm setup with several printers connected to a single PC.

Sorry it was early when I replied this morning but when I suggested adding a .local I meant to your hosts files or wherever docker networks manage all that. I also meant any extension on the domain, doesn't have to be .local. Is .local specific to mDNS? I would have thought you're host file would override any external DNS anyway. I forget my networking basics aha. 

Pretty mute point to make anyway with your other work around working, but thought I'd mention it incase anyone else stumbles along this issue. 

There isn't really a best practice for 3d printer farms. You've got a massive mix of what people actually do and to be totally honest having OctoPrint sat as an extra bit of hardware isn't really best practice for larger farms i.m.o as it's additional points of failure I'd rather avoid, especially with relying on USB! 

I'd go with what you know personally :), I've seen people try to learn kube just to run a farm with all this fancy redundancy when it takes 10-15 minutes to flash a new pi.... 

I've tried VMs with multiple USB PCI-E cards attached, Pi per printer (not worth the hassil, especially now), Pi per 4 printers (only good on power draw as even Pi add up, limited by physical USB's. as Pi4 runs more than 4 OP instances easily)

The main thing people try to go for is redundancy, so they'd rather not have 1 Pi/Host with multiple OP instances running on them, as that host could in theory take out multiple printers. I had 5 printers running at home at one point, all running from a single VM as my server was in USB reach of each and I had that happen to me twice in 3 years, both was me been a dodo but I had the VM running on my media server which I always tinker with.

Daghis, 2022-11-18:
Thank you very much, @NotExpectedYet for the very detailed and well-considered reply. My experience with Docker is only slightly better than a casual user level, but I'm reasonably pleased with the little setup I have (formerly a gaming rig with 2 7-port USB hubs, one on each rack), running Docker containers for OctoFarm and all the OctoPrint instances (plus one for Portainer because I generally do better with a GUI 😄).

Although `.local` would likely just well enough as we each said, [here's a reference](https://en.wikipedia.org/wiki/.local) to its "special-use"ness, in case you were curious.

NotExpectedYet, 2022-11-18:
> Thank you very much, @NotExpectedYet for the very detailed and well-considered reply. My experience with Docker is only slightly better than a casual user level, but I'm reasonably pleased with the little setup I have (formerly a gaming rig with 2 7-port USB hubs, one on each rack), running Docker containers for OctoFarm and all the OctoPrint instances (plus one for Portainer because I generally do better with a GUI smile).
> 
> Although `.local` would likely just well enough as we each said, [here's a reference](https://en.wikipedia.org/wiki/.local) to its "special-use"ness, in case you were curious.

Your most welcome! I think your setup sounds fine personally and your running pretty close to how I test so you'll have the most tested configuration aha. 

I will certainly give that a read!