OctoFarm/OctoFarm #1336: I dont remember the username I entered. How do you remove users?

tablifiedapps, 2022-11-21:
### OctoFarm Version

latest 1.7

### OctoPrint Version(s)

1.8

### pm2 or docker?

pm2

### Operating System

ubuntu

### Description of the problem

I wasnt paying attention after I installed octofarm on my ubuntu server. When I got to the creation screen I entered info but don't remember actually entering a username that I wanted. I put all my info like my name, email, etc and a password. But now it is asking for the username and I tried all the common ones I would use but it's not letting me in. Theres no way to reset anything.

### Steps taken to reproduce

1. Tried deleting the octofarm folder and pulling it again but the login info is stored in a db somewhere and I don't know where it is

NotExpectedYet, 2022-11-21:
There's a .env file inside your OctoFarm folder, append a `2` or something on the end of the database uri field

tablifiedapps, 2022-11-22:
Thank you so much! This fixed it.

NotExpectedYet, 2022-11-22:
> Thank you so much! This fixed it.

Welcome, I'll leave this ticket open until I've added it to the docs :)