OctoFarm/OctoFarm #1338: History resets itself and shows 1 day Total Print Time

HappyOldtimer, 2022-12-01:
### OctoFarm Version

1.7.3

### OctoPrint Version(s)

1.8.6

### pm2 or docker?

pm2

### Operating System

Ubuntu 20.04

### Description of the problem

The History showed, well, the history of the prints. However, upon starting my desktop computer and opening the Octofarm site in the browser, it only showed 1 day as the number of Total Print Days. 

### Steps taken to reproduce

Good question, I am unsure, sorry.

(Apologies in advance if I didn't submit this issue correctly, this is the first time I've ever done this)

[octofarm_dump-12-1-2022-9-56-26 AM.zip](https://github.com/OctoFarm/OctoFarm/files/10131292/octofarm_dump-12-1-2022-9-56-26.AM.zip)
[image]
[image]

NotExpectedYet, 2022-12-01:
Hello, thanks for reporting the issue! 

You filled it in perfect, even gave me logs so thanks again :D

Going from what you've told me I believe it's because the initial run for generating the history graph but that seems to have completed succesfully in the logs. 

The data is certainly still there and will be collecting, just looks like a UI bug, or something hidden in the generating. 

I'd be curious to know if it recovers itself after another print is captured? (that re-forces the generation) or another reboot/restart? unless you've already experience/tried those. 

Something certainly going awry with the code though by the looks. 

Just so I can rule out some previous changes, could you let me know if you used the Filament Manager plugin on your instances before? I've had some issues crop up from removing that and would just like to rule it out.

HappyOldtimer, 2022-12-01:
Hi James, glad to hear I included everything needed.
(and just for the record: I think OctoFarm is just awesome :-)

Yes, I would concur that it seems a UI glitch/issue, as the data still seems to be there in for example the dashboard graph 'completion by day', and in the list of prints in the history section. 
(As an aside: It seems though that several other dashboard UI panels also don't look back to the start date?) 

I have used the Filament Manager as it is in there from the install of OctoFarm several weeks ago, I do not recall installing that separately, as far as I know it just came with OctoFarm. I have not installed nor used a Filament manager plugin on the OctoPrint instances, that I am 100% sure of. 

And I have just harvested two prints, but History is unfortunately not resetting itself.

Regards,
Paul

NotExpectedYet, 2022-12-01:
> Hi James, glad to hear I included everything needed. (and just for the record: I think OctoFarm is just awesome :-)
>
> Yes, I would concur that it seems a UI glitch/issue, as the data still seems to [...]

That perfect thank you very much Paul, cleared that up. Also thank you for the praise, I've not had much time to put work into it recently due to life but I am glad people still get a use out of it. 

Narrows it down at least, I'll investigate the code and see if I can figure out why. This issue will automatically close when the release hits with the next version. I'm not sure when that be sadly but will certainly fix it. 

I dunno how attached to your history you are but there's some database functions in System -> Database that will allow you to nuke the whole history database and leave other parts of it. That would clear out whatever record is causing the issue so thought I'd mention it.

NotExpectedYet, 2022-12-08:
Hi Paul, Must be some other bug then from what you've said. Would you mind sending through another log dump please? A clean set from the nuke might help allude to something more.  

I can see those spools been saved in your previous log at least

```
"filamentSelection":[{"spools":{"name":"Pistachio Green","profile":{"index":null,"density":1.27,"diameter":1.75,"manufacturer":"Prusa","material":"petg"},"price":22.31,"weight":1000,"used":1005.7980551862639,"tempOffset":0,"bedOffset":0,"fmID":null},
```

HappyOldtimer, 2022-12-09:
Hi James,

Sure, no worries, see attached :-)
Just to make sure though: some rolls were assigned to printers, but still showed up in History as '0: no spool'. 

[octofarm_dump-12-9-2022-10-44-16 AM.zip](https://github.com/OctoFarm/OctoFarm/files/10193982/octofarm_dump-12-9-2022-10-44-16.AM.zip)