OctoFarm/OctoFarm #1344: Error while looking for printer info after completed print

losless50, 2022-12-09:
### OctoFarm Version

monolithic-latest 

### OctoPrint Version(s)

Latest 

### pm2 or docker?

docker

### Operating System

Ubuntu 22.04

### Description of the problem

When I click the info button in the panel view to see my material/maintenance costs it's stuck on "loading" if this an issue on my end? and how can I fix it? BTW Love octofarm! amazing work so far!
[image]

### Steps taken to reproduce

open the panel view
click info button on any printer during or after print

losless50, 2022-12-09:
[_octofarm-monolithic_logs.txt](https://github.com/OctoFarm/OctoFarm/files/10197776/_octofarm-monolithic_logs.txt)

NotExpectedYet, 2022-12-09:
Hello thanks for the report! 

It's a bug in the client code I will have to fix, it's trying to read a property of a value that doesn't exist and javascript throws a wobbler with that.

NotExpectedYet, 2022-12-09:
Also thanks for the praise!

losless50, 2022-12-09:
Okay awesome! I was worried something was broken on my end, thank you so much! is there a version I could revert back to until this issue is resolved?