OctoFarm/OctoFarm #1351: SELinux Support

aardvarkglory, 2022-12-16:
### OctoFarm Version

1.7.3

### OctoPrint Version(s)

1.8.6

### pm2 or docker?

pm2

### Operating System

RHEL, CentOS, Rocky Linux, Oracle Linux, AlmaLinux, Fedora

### Description of the problem

PM2 restarts OctoFarm every 90 seconds due to SELinux.

### Steps taken to reproduce

1. OctoFarm and PM2 work perfect when SELinux is disabled.
2. Also works by editing the OctoFarm PM2 systemd file. (You will need to change the username for the PM2 service.)
    sudo vim /etc/systemd/system/pm2-username.service

Add the following line:
Environment=PM2_PID_FILE_PATH=/run/pm2.pid

Replace: PIDFile=/username/.pm2/pm2.pid to: PIDFile=/run/pm2.pid

I am not sure how to make SELinux ignore the PM2 service and OctoFarm home folder.

[image]

NotExpectedYet, 2022-12-17:
I believe something like this should work. It will however disable the internal updater/restarting mechanism. 
```
[Unit]
Description=OctoFarm Service
Documentation=https://docs.octofarm.net
After=network.target

[Service]
Type=simple
User={home_user}
ExecStart=/usr/bin/node /home/{home_user}/OctoFarm/server/app.js
Restart=on-failure

[Install]
WantedBy=multi-user.target
```

NotExpectedYet, 2022-12-17:
I think, as mentioned in the discussion, adding support for custom restart commands and allowing those to be added to an environment file would help. 

pm2 was there just so people didn't have to mess about in systemd but no reason advanced users should be limited to pm2.

aardvarkglory, 2022-12-21:
> I believe something like this should work for systemd if it helps. It will however disable the internal updater/restarting mechanism.
> 
> ```
> [Unit]
> Description=OctoFarm Service
> Documentation=https://docs.octofarm.net
> After=network.target
> 
> [Service]
> Type=simple
> User={home_user}
> ExecStart=/usr/bin/node /home/{home_user}/OctoFarm/server/app.js
> Restart=on-failure
> 
> [Install]
> WantedBy=multi-user.target
> ```

I tried this and it did not work.  The best way is as follows

Check to see if OctoFarm is running PM2: `pm2 list`.  If OctoFarm is not running, start it and then check again.

Once OctoFarm is running, run the following command to add PM2/OctoFarm to systemd: `sudo env PATH=$PATH:/usr/bin /usr/lib/node_modules/pm2/bin/pm2 startup systemd -u {home_user} --hp /home/{home_user}`.

Edit the OctoFarm PM2 systemd file. (You will need to change the username for the PM2 service.)
`sudo vim /etc/systemd/system/pm2-{home_user}.service`
Add the following line:
`Environment=PM2_PID_FILE_PATH=/run/pm2.pid`

Replace: `PIDFile=/{home_user}/.pm2/pm2.pid` to: `PIDFile=/run/pm2.pid`