OctoFarm/OctoFarm #1372: 404 error when uploading files from OctoFarm to Octoprint

dpvdberg, 2023-01-07:
### OctoFarm Version

1.7.3

### OctoPrint Version(s)

1.8.6

### pm2 or docker?

docker

### Operating System

Ubuntu

### Description of the problem

Whenever I try to upload a file from OctoFarm to OctoPrint, it gives me the following error:

[image]

When I check with Fiddler, I get a failed post to `/octoprint/63b4c057d820aa715ed67a5f/api/files/local`
Body is the file to be uploaded, response is
`
HTTP/1.1 404 Not Found
x-ratelimit-limit: 10000
x-ratelimit-remaining: 9998
date: Sat, 07 Jan 2023 00:51:16 GMT
x-ratelimit-reset: 1673052679
x-dns-prefetch-control: off
expect-ct: max-age=0
strict-transport-security: max-age=15552000; includeSubDomains
x-download-options: noopen
x-content-type-options: nosniff
origin-agent-cluster: ?1
x-permitted-cross-domain-policies: none
referrer-policy: no-referrer
x-xss-protection: 0
content-security-policy: default-src 'none'
content-type: text/html; charset=utf-8
content-length: 155
keep-alive: timeout=5

<!DOCTYPE html>
<html lang="en">
<head>
<meta charset="utf-8">
<title>Error</title>
</head>
<body>
<pre>Cannot POST /api/files/local</pre>
</body>
</html>
`

I have verified that the user has correct permissions and CORS is enabled.
Also note, when I go to the URL in my browser, I get a JSON of files that are present on the OctoPrint instance, so that seems to work...

### Steps taken to reproduce

1. Create Docker.
2. Add OctoPrint instance.
3. Try to upload file.

NotExpectedYet, 2023-01-07:
Thanks for the report. 

Hmm seems like the files going through. CORS isn't needed anymore as files go from server to server not browser to server. Seems like something on the server is getting confused. 

Could you pass me a log dump from the system page please?

dpvdberg, 2023-01-07:
I was using Octoprint behind a reverse proxy (<printername>.<domain>.com).
However, when I use the internal LAN ip address for each Octoprint instance, it seems to work fine :)

Thanks anyway!

closed 2023-01-07 by dpvdberg

NotExpectedYet, 2023-01-07:
> I was using Octoprint behind a reverse proxy (_printername_._domain_.com). However, when I use the internal LAN ip address for each Octoprint instance, it seems to work fine :)
> 
> Thanks anyway!

Your welcome! That might be to do with your docker setup and routing as OctoFarm supports hosts behind a reverse proxy. The only thing it seems to have trouble with is mDNS resolutions. 

All connections are proxied through OctoFarm now so it's better to connect directly anyway.