OctoFarm/OctoFarm #1377: Accidentally updated nodejs via ubuntu manager, OctoFarm fails to start due to nodejs error ".bin is a directory"

tablifiedapps, 2023-01-15:
### OctoFarm Version

1.7

### OctoPrint Version(s)

1.8.6

### pm2 or docker?

pm2

### Operating System

Ubuntu

### Description of the problem

expected to be able to run npm start and octofarm will work

### Steps taken to reproduce

I cant recall exactly what I updated. I saw the update pop-up on ubuntu to update various apps and I did see node being listed.

After installing, I noticed I couldnt access octofarm. My printers are still accessible individually.

I run the npm start after cd into my octofarm folder and I get this error (log attached)
[2023-01-15T22_03_43_861Z-debug-0.log](https://github.com/OctoFarm/OctoFarm/files/10421383/2023-01-15T22_03_43_861Z-debug-0.log)

I tried uninstalling nodejs
Reinstalling nodejs
I ran the commands from setp 2.3 to 2.5 again from this site
https://docs.octofarm.net/installation/install-linux-ubuntu.html#_2-4-update-npm-to-the-latest

Nothing seems to work.

tablifiedapps, 2023-01-16:
Other things I have tried is deleting the octofarm folder and installing it again via git. The interesting thing is that it worked. I was able to load octofarm. but then my credentials weren't working. so I renamed the env file and restarted octofarm then go the same error as earlier.

NotExpectedYet, 2023-01-16:
For some reason the .log won't open on my phone. I'll be near a PC tomorrow morning for me if you can wait. Should be fixable. :) Alternatively if you can upload a screenshot with the outputs of the below that would help diagnose the issue. 

`npm -v`
`node -v` 

Inside your OctoFarm folder 

`npm start` - just the bit where it fails.

sfjelstad, 2023-01-17:
I did a bit of digging since I was having a similar issue and it looks like the culprit is actually ```npm ci``` being run in ```server/package.json```. Turns out it looks like ```npm ci``` was actually broken upstream and a release with the patch hasn't been released yet. For the moment manually removing ```server/node_modules``` prior to running should fix it, or alternatively you could switch back to an older version of npm using ```npm install -g npm@9.2.0```

https://github.com/npm/cli/issues/6051

NotExpectedYet, 2023-01-17:
Huh nice, thanks for figuring that out. How annoying. I'll keep this ticket open until Npm has resolved at least others can find the same resolution. 

Things like this are giving me more gusto to tackling packaging up a binary at least.

sfjelstad, 2023-01-18:
Quick update, npm version 9.3.1 has been released and the fix appears to be implemented in the patch notes. ```npm install -g npm@latest``` should work again.

https://github.com/npm/cli/releases/tag/v9.3.1

NotExpectedYet, 2023-01-18:
Brill thank you for letting me know

tablifiedapps, 2023-01-19:
Thanks for this! I feel like I wont update anything until I check on github.

However, I am still having issues. I didnt get any errors in my terminal but when I go to the URL I get a page not found error. Octoprints are still working individually. I made sure the mongodb is started as well. I feel like I messed something up in my haste to fix it.

tablifiedapps, 2023-01-19:
klippy@klipbox:~/OctoFarm$ npm start

> octofarm@1.7.3 start
> cd server && npm run start

> octofarm-server@1.7.3 prestart
> npm ci

added 760 packages, and audited 761 packages in 18s

146 packages are looking for funding
  run `npm fund` for details

6 vulnerabilities (2 moderate, 4 high)

To address issues that do not require attention, run:
  npm audit fix

To address all issues possible (including breaking changes), run:
  npm audit fix --force

Some issues need review, and may require choosing
a different dependency.

Run `npm audit` for details.

> octofarm-server@1.7.3 start
> pm2 flush && pm2 start ecosystem.config.js

[PM2] Flushing /home/klippy/.pm2/pm2.log
[PM2] Flushing:
[PM2] /home/klippy/OctoFarm/logs/pm2.out.log
[PM2] /home/klippy/OctoFarm/logs/pm2.error.log
node:internal/fs/utils:348
    throw err;
    ^

Error: ENOENT: no such file or directory, open '/home/klippy/OctoFarm/logs/pm2.out.log'
    at Object.openSync (node:fs:600:3)
    at /usr/local/lib/node_modules/pm2/lib/API/LogManagement.js:41:27
    at Array.forEach (<anonymous>)
    at /usr/local/lib/node_modules/pm2/lib/API/LogManagement.js:31:12
    at /usr/local/lib/node_modules/pm2/node_modules/pm2-axon-rpc/lib/client.js:45:10
    at Parser.<anonymous> (/usr/local/lib/node_modules/pm2/node_modules/pm2-axon/lib/sockets/req.js:67:8)
    at Parser.emit (node:events:513:28)
    at Parser._write (/usr/local/lib/node_modules/pm2/node_modules/amp/lib/stream.js:91:16)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10) {
  errno: -2,
  syscall: 'open',
  code: 'ENOENT',
  path: '/home/klippy/OctoFarm/logs/pm2.out.log'
}

Node.js v18.13.0

NotExpectedYet, 2023-01-20:
For 1.8.0 I'm going to be putting in a notice to the Systems page, as well as update notices/change logs to try and help mitigate stuff like this in the future. 

Sadly there's nothing I can really do about the underlying systems those will need maintaining by the user but I'm working on packaging up this application into a binary for 2.0 so will remove a lot of the required tooling.

NotExpectedYet, 2023-01-20:
Closing this as the original issue with npm has been resolved and is live now.

closed 2023-01-20 by NotExpectedYet