OctoFarm/OctoFarm #1380: Docker webpage octofarm not accessible after creation

Doprintityourself, 2023-01-31:
### OctoFarm Version

latest

### OctoPrint Version(s)

latest

### pm2 or docker?

pm2

### Operating System

Linux Docker

### Description of the problem

after composing a stack in Portainer both containers are working but website of Octofarm ist not reachable on port 4000
Attached the log file of the Octofarm container there are errors
As I am not a Programmer I cannot understand or fix

Thanks for having a look

### Steps taken to reproduce

just  composed the script provided on the Octofarm docker dokumentation replaced admin an password as required:

version: '3.4' 

services:
  mongodb:
    image: mongo:latest
    container_name: mongodb
    environment:
      MONGO_INITDB_ROOT_USERNAME: admin
      MONGO_INITDB_ROOT_PASSWORD: mypw
      MONGO_INITDB_DATABASE: octofarm
    volumes:
    # Local volume (change to mongodb-data for a named volume folder)
      - ./mongodb-data:/data/db
    restart: unless-stopped

  octofarm:
    container_name: octofarm
    # choose octofarm/octofarm:latest
    image: octofarm/octofarm:latest
    restart: unless-stopped
    links:
      - mongodb
    ports:
      - 4000:4000 # port of SYSTEM : port of CONTAINER
    environment:
      - MONGO=mongodb://admin:mypw@mongodb:27017/octofarm?authSource=admin
    volumes:
    # Volumes as local relative folders (validate with 'docker-compose config')
      - ./OctoFarm/logs:/app/logs
      - ./OctoFarm/scripts:/app/scripts
      - ./OctoFarm/images:/app/images
log: 
👍 2023-01-31T09:29:20.424987161Z 09:29:20 0|OctoFarm  | 2023-01-31T09:29:20: 1/31/23, 9:29:20 AM | INFO | Server-Env  
2023-01-31T09:29:20.425000988Z 09:29:20 0|OctoFarm  | 2023-01-31T09:29:20:  INFO MESSAGE: ✓ Running OctoFarm version 1.7.3 in non-NPM mode! 
2023-01-31T09:29:20.425360067Z 09:29:20 0|OctoFarm  | 2023-01-31T09:29:20: 1/31/23, 9:29:20 AM | INFO | Server-Env  
2023-01-31T09:29:20.425400293Z 09:29:20 0|OctoFarm  | 2023-01-31T09:29:20:  INFO MESSAGE: ✓ MONGO environment variable set! 
2023-01-31T09:29:20.425884969Z 09:29:20 0|OctoFarm  | 2023-01-31T09:29:20: 1/31/23, 9:29:20 AM | WARN | Server-Env  
2023-01-31T09:29:20.425921103Z 09:29:20 0|OctoFarm  | 2023-01-31T09:29:20:  WARN MESSAGE: ~ The OCTOFARM_PORT setting was not a correct port number: >= 0 and < 65536. Actual value: undefined. 
2023-01-31T09:29:20.426628673Z 09:29:20 0|OctoFarm  | 2023-01-31T09:29:20: 1/31/23, 9:29:20 AM | ERROR | Utils-Env  
2023-01-31T09:29:20.426640941Z 09:29:20 0|OctoFarm  | 2023-01-31T09:29:20:  ERROR MESSAGE: cant find target folder... | /app/images
2023-01-31T09:29:20.431216177Z 09:29:20 0|OctoFarm  | 2023-01-31T09:29:20: Error: EACCES: permission denied, open '../images/bg.jpg'
2023-01-31T09:29:20.431244774Z 09:29:20 0|OctoFarm  | 2023-01-31T09:29:20:     at Object.openSync (fs.js:498:3)
[... 65 lines cut ...]
2023-01-31T09:29:52.001824112Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:  INFO MESSAGE: ✓ MONGO environment variable set! 
2023-01-31T09:29:52.002258893Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52: 1/31/23, 9:29:52 AM | WARN | Server-Env  
2023-01-31T09:29:52.002299736Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:  WARN MESSAGE: ~ The OCTOFARM_PORT setting was not a correct port number: >= 0 and < 65536. Actual value: undefined. 
2023-01-31T09:29:52.002996078Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52: 1/31/23, 9:29:52 AM | ERROR | Utils-Env  
2023-01-31T09:29:52.003050154Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:  ERROR MESSAGE: cant find target folder... | /app/images
2023-01-31T09:29:52.007816418Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52: Error: EACCES: permission denied, open '../images/bg.jpg'
2023-01-31T09:29:52.007863303Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:     at Object.openSync (fs.js:498:3)
2023-01-31T09:29:52.007884924Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:     at Object.writeFileSync (fs.js:1529:35)
2023-01-31T09:29:52.007889173Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:     at Object.ensureBackgroundImageExists (/app/server/utils/env.utils.js:112:8)
2023-01-31T09:29:52.007893670Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:     at setupEnvConfig (/app/server/app-env.js:282:12)
2023-01-31T09:29:52.007903638Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:     at Object.<anonymous> (/app/server/app.js:27:3)
2023-01-31T09:29:52.007938181Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:     at Module._compile (internal/modules/cjs/loader.js:1085:14)
2023-01-31T09:29:52.007942886Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
2023-01-31T09:29:52.007956072Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:     at Module.load (internal/modules/cjs/loader.js:950:32)
2023-01-31T09:29:52.007979026Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:     at Function.Module._load (internal/modules/cjs/loader.js:790:12)
2023-01-31T09:29:52.007997402Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:     at Object.<anonymous> (/usr/local/lib/node_modules/pm2/lib/ProcessContainerFork.js:33:23) {
2023-01-31T09:29:52.008021570Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:   errno: -13,
2023-01-31T09:29:52.008060760Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:   syscall: 'open',
2023-01-31T09:29:52.008065696Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:   code: 'EACCES',
2023-01-31T09:29:52.008084744Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52:   path: '../images/bg.jpg'
2023-01-31T09:29:52.008102109Z 09:29:52 0|OctoFarm  | 2023-01-31T09:29:52: }
2023-01-31T09:29:52.013813746Z 2023-01-31T09:29:52: PM2 log: App [OctoFarm:0] exited with code [1] via signal [SIGINT]
2023-01-31T09:29:52.014134044Z 2023-01-31T09:29:52: PM2 log: App [OctoFarm:0] will restart in 15000ms
2023-01-31T09:29:52.014381004Z 09:29:52 PM2         | App [OctoFarm:0] exited with code [1] via signal [SIGINT]
2023-01-31T09:29:52.014565974Z 09:29:52 PM2         | App [OctoFarm:0] will restart in 15000ms

NotExpectedYet, 2023-01-31:
It's a permissions issue, you'll need to `sudo chmod 777 -R octofarm/` or whatever your octofarm directory is. 

I'm planning to fix the docker to make this not required but I'm half way through moving house currently and don't have access to a PC

Doprintityourself, 2023-01-31:
sorry to ask but may be you can help
how do I access octofarm container seems not working: 
1. docker exec -it octofarm /bin/bash is not working  either in cli nore in portainer
2. Is there a password to access the container

NotExpectedYet, 2023-01-31:
> sorry to ask but may be you can help how do I access octofarm container seems not working:
> 
> 1. docker exec -it octofarm /bin/bash is not working  either in cli nore in portainer
> 2. Is there a password to access the container

it's outside the container, on your host. I assume is Linux?

Doprintityourself, 2023-01-31:
As I created it inside portainer I assume it is residing inside a docker container and not on teh filesystem. both mogdb and octofarm are in containers  
find -name octofarm does not give any results on the linux filesystem. 
So isent it created in the octofarm container/encrypted?
I am not familiar what a container does or is or if there inside is the octofarm directory structure.

NotExpectedYet, 2023-01-31:
And what system is portainer installed on? 

Dockers are finite so they require access to the local file system. You've defined mount paths above as `./OctoFarm/**` that path need permissions changing for it to work.

NotExpectedYet, 2023-01-31:
I recommend you do a bit more reading about dockers and how they work. It will help a lot with troubleshooting steps.

Doprintityourself, 2023-02-01:
I composed it by hand again on cli and set the permissions
I can reach the Octofarm url now and as it seems it now hast created real octofarm directory that I have given rights.

Obviously something different happend or I should say not happend during the stack creation with the same script.
It seems to build the mogo_db and the octofarm container but not the real structure via downlaod a suppose.
Thanks inspite of that for your support
have nice day