OctoFarm/OctoFarm #1386: OctoFarm 1.7.3 is sluggish, crashes, loses connection to OctoPrint

ndarrow, 2023-03-23:
### OctoFarm Version

1.7.3

### OctoPrint Version(s)

1.8.2

### pm2 or docker?

pm2

### Operating System

Linux 5.4.0-1081-raspi (linux)

### Description of the problem

Sluggish-
Over time, all inputs are very slow to execute. Example- pressing the print button from List view can take several minutes to send the print request (I've watched OP at the same time and can see the delay). This goes for all user inputs- changing views, selecting files, starting/stopping/pausing prints... multi print can take 15+ minutes to complete (if it doesn't crash). 

Crashing-
OF will often crash, usually preceded by sluggish behavior. Sometimes it will stop responding and eventually show the browser's "Unresponsive" sad face. Other times the OF "lost connection" pop up will appear. After crashes, OF will sometimes recover and come back (usually it's so slow after this I need to force a reboot). Sometimes it remains unresponsive and I can SSH > sudo reboot; other times SSH locks at the user/PW, sometimes the SSH session is established, but there's no user/PW prompt in the terminal- just black screen. 

Losing connections to OctoPrint-
OF will randomly lose connections to OP instances while the OP instances seem to be running properly (I can log into OP with no problems and it is snappy and responsive). Sometimes just one printer, sometimes the whole farm and everything in between. I usually have to go to the "Printers" tab and select "Force Reconnect" to reconnect. Sometimes cycling power on the OP pi is the only way to reconnect (even though I can log into the OP instance). 

I have 54 instances of OP on 54 devices (raspberry pis, Panda boards, octo4a/phone). All 54 devices are using WiFi. 
OF is on a raspberry pi 4 with its own dedicated 120vac desk fan. 

### Steps taken to reproduce

[octofarm_dump-3-23-2023-2-12-08 AM.zip](https://github.com/OctoFarm/OctoFarm/files/11046435/octofarm_dump-3-23-2023-2-12-08.AM.zip)

I got this right before a crash. I think half the farm lost connection right before I grabbed the logs. 

[image]

[image]
Not sure why there's always so many active sessions... I'm only logged in on one device in this screenshot.

ndarrow, 2023-03-23:
Also, sometimes when printers lose connection, they will randomly be kicked from List view and won't appear again until communication is restored. This happens even though I have "Offline Printers" and "Disconnected Printers" checked in the Views Visibility section. 

[image]

NotExpectedYet, 2023-03-23:
Regarding the multiple Active Sessions, the implementation for the websockets isn't unified across pages so every page load is a new session and the old ones probably don't time out for a bit. That has been resolved in 1.8.0 :)

NotExpectedYet, 2023-03-23:
I believe there's a memory leak here and I'm not seeing it on my work instance because there's only 3 instances running.

NotExpectedYet, 2023-03-23:
Yep confirmed, all the v8 heap crashes are in the logs. Along with a few other bugs. :/

ndarrow, 2023-03-23:
Not sure if this is helpful, but here's a log dump taken while things are working well and all printers printing. This was pretty soon after I rebooted OF. 

[octofarm_dump-3-23-2023-2-53-50 PM.zip](https://github.com/OctoFarm/OctoFarm/files/11052425/octofarm_dump-3-23-2023-2-53-50.PM.zip)