OctoFarm/OctoFarm #1387: File Manager is not accessible

rajavel1990, 2023-03-23:
### OctoFarm Version

1.7.3

### OctoPrint Version(s)

1.8.6

### pm2 or docker?

pm2

### Operating System

Windows 11

### Description of the problem

when i try to open the file manager to access the files this error is encountered

[image]

### Steps taken to reproduce

on Panel View
When clicked on the Files Button on a Printer name 
the error is triggered and no files are displayed

rajavel1990, 2023-03-23:
I Have been running without any issues for the last 5 months on Windows 11 but suddenly i am facing this issues. I deleted the install folder and reinstalled again but still the same issue

[image]

NotExpectedYet, 2023-03-23:
After a windows 11 update by anychance? 

There's something wrong between the client communication and the server thats running. The 400 requests indicates the server cannot process what the client is sending. 

Can you grab them out of the folder you've installed OctoFarm into? 

At a complete guess is the nodejs process has lost permissions to your folder but that's a guess from a previous situation.

rajavel1990, 2023-03-23:
I have restarted the connected computer that has the octofarm installed and now it is working as normal.,

But there is a problem with a few printers that are connected through Raspberry pi.

the octofarm does show the added printers on the dashboard. When it is printing it does show the print progress and i am able to restart the print from the dashboard. But the control tab does not work none of the commands are executed.

It does not show the tool temperature and the printer controls are non functional. I have attached the screenshot along with this

[image]

easonjim, 2023-04-25:
This question is tested:
1. It is not a domain name, nor a reverse proxy problem
2. When it is node proxy, the obtained body is empty
3. It has nothing to do with the node version, whether it is 14 or 18, this problem also exists
4. It has nothing to do with the system, whether it is windows, linux, or mac, the same problem exists

It is currently being repaired, please pay attention to pr later.

easonjim, 2024-01-16:
After several months of investigation, I found that it was not a code bug that affected this function, but an incomplete dependency package.
After I re-deployed on Linux and re-installed the dependent packages using npm, the problem was solved.