OctoFarm/OctoFarm #1389: Looking for maintaners

NotExpectedYet, 2023-04-11:


wondertory, 2023-04-15:
I can pick it up and run with it.

NotExpectedYet, 2023-04-18:
> I can pick it up and run with it.

Awesome! 

I'm pretty happy to add you to the organisation as a maintainer. 

Would you mind getting a PR ready for some work you'd like to do? 

I just want to make sure it's passed over to someone willing and able :)

easonjim, 2023-04-19:
May I join this repository for maintenance?

VedanthB, 2023-04-20:
Hey @NotExpectedYet I am a frontend developer and I have some experience contributing to open-source projects. I have always been passionate and excited about 3D printing and I believe I can make a positive contribution to the OctoFarm community and learn a ton from maintaining this platform.

Please let me know if there is any further information I can provide or if you have any questions for me. I am looking forward to the opportunity to be a part of this project.

mswdev, 2023-04-24:
@NotExpectedYet Are you willing to let new maintainers start off with the v2 codebase you were working on? It sounds like that was further along and could be a good starting point.

I'd be willing to potentially join as a maintainer, but I'd likely fork the project if you're moving on permanently.

NotExpectedYet, 2023-04-24:
Hello everyone, 

It was my intention to pass this repository over to whomever wanted to take over but I think it probably better for someone to create a fork and continue on. 

My reasoning for keeping the work on this repo has gone now as I don't think I'll be returning to the project so I will probably just archive it when I get chance and put a note on the README. 

If anyone does create a fork and would like a link back from this repo please do let me know and I will do that!  

@mswdev the V2 code base isn't that far along, the majority of it was still in the planning stages. The most I'd gotten written was the server initialisation into dependency injection. I was mostly trying to figure out how to make the printer connector a configuration file, so anyone could create a new printer connector. The rest was upgrading bootstrap, which I've already ported most of the V2 UI work I'd done into the UI branch here but there was a lot still to continue on that, upgrading bootstrap killed a lot of stuff.

mswdev, 2023-04-26:
@NotExpectedYet I send you over a DM on Discord.

NotExpectedYet, 2023-04-29:
@easonjim @VedanthB @wondertory 

Sorry about my tardy reply, I think the more maintainers the better personally but I have left the repository in @mswdev's hands. He seemed as passionate as me in discussions. I've no qualms with you two been on the maintainer team as well if you two are still game for it? I am about today on github so if you drop me a message I shall enable that as well. 

I've left mswdev as Administrator/Owner level of the organisation currently.

orangeworx, 2023-05-13:
sign me up :D

mswdev, 2023-05-29:
@NotExpectedYet I've been trying to reach you via Discord, but I've not been able to get ahold of you. Could you get back to me regarding our discussions?

emersonsc, 2023-08-14:
> @NotExpectedYet I've been trying to reach you via Discord, but I've not been able to get ahold of you. Could you get back to me regarding our discussions?

Looks like he's been in-active on github for some time. Maybe start a fork? I have a buddy who uses this and would love to see this go forward and maybe get some Moonraker support for my farm.

T3chGuy77, 2024-06-24:
Seems this project is dead now. Sad the original developer never did as he said he'd do and let someone take over the project. Seems there was interest from several individuals, the more the better. I can't find any forks either. Seemed like a great project too. Here's to hoping it gets forked.

gmccauley, 2024-06-25:
> Seems this project is dead now. Sad the original developer never did as he said he'd do and let someone take over the project. Seems there was interest from several individuals, the more the better. I can't find any forks either. Seemed like a great project too. Here's to hoping it gets forked.

FDM-Monster is a fork of OctoFarm.

https://github.com/fdm-monster/fdm-monster

I can't speak to how active the development is or how well it works.  I never tried it.  I switched over and started using Prusa Connect instead.