OctoFarm/OctoFarm #1393: Server not starting

orangeworx, 2023-05-13:
### OctoFarm Version

1.7.3

### OctoPrint Version(s)

1.8.6

### pm2 or docker?

pm2

### Operating System

RPi - FarmPi 0.1.0

### Description of the problem

Can't get the mongo server started.

pm2.error.log
2023-05-12T23:56:55: MongoServerSelectionError: connect ECONNREFUSED 127.0.0.1:27017
2023-05-12T23:56:55:     at Timeout._onTimeout (/home/pi/OctoFarm/server/node_modules/mongodb/lib/sdam/topology.js:293:38)
2023-05-12T23:56:55:     at listOnTimeout (internal/timers.js:557:17)
2023-05-12T23:56:55:     at processTimers (internal/timers.js:500:7) {
2023-05-12T23:56:55:   reason: TopologyDescription {
2023-05-12T23:56:55:     type: 'Unknown',
2023-05-12T23:56:55:     servers: Map(1) { '127.0.0.1:27017' => [ServerDescription] },
2023-05-12T23:56:55:     stale: false,
2023-05-12T23:56:55:     compatible: true,
2023-05-12T23:56:55:     heartbeatFrequencyMS: 10000,
2023-05-12T23:56:55:     localThresholdMS: 15,
2023-05-12T23:56:55:     setName: null,
2023-05-12T23:56:55:     maxElectionId: null,
2023-05-12T23:56:55:     maxSetVersion: null,
2023-05-12T23:56:55:     commonWireVersion: 0,
2023-05-12T23:56:55:     logicalSessionTimeoutMinutes: null
2023-05-12T23:56:55:   },
2023-05-12T23:56:55:   code: undefined,
2023-05-12T23:56:55:   [Symbol(errorLabels)]: Set(0) {}
2023-05-12T23:56:55: }

### Steps taken to reproduce

I was running an update. It completed without issues but after reboot, I couldn't load OctoFarm, just a webpage with this message
"The OctoFarm server is currently not running
If you just started up your Raspberry Pi, please wait a couple of seconds, then try to refresh this page.

If the issue persists, please log into your Raspberry Pi via SSH and check the following:

Verify that the process is running: ps -ef | grep -i OctoFarm | grep -i node should show a node process:
pi@farmpi:~ $ ps -ef | grep -i OctoFarm | grep -i node
pi 1441 1 6 11:12 ? 00:00:15 node /home/pi/OctoFarm/app.js
If it isn't, the question is why. Take a look into ~/.OctoFarm/logs/pm2.error.log, there might be an error logged in there that helps to determine what's wrong.
You might also want to try if you can restart the server (if no obvious error is visible): sudo service pm2-pi restart.
If all that doesn't help to trouble shoot the issue, you can seek support on the [OctoFarm Discord](https://discord.octofarm.net/) or [OctoFarm Subreddit](https://reddit.com/r/OctoFarm). Please provide your FarmPi and OctoFarm versions and explain what you already tried and observed as detailed as possible."

I'm stuck at if it isn't. It is there. The problem has to do the mongodb server not being started and now wanting to be started.
Tried difference things including updating mongo to 6.0

Any ideas?

Dumnersm580, 2023-05-13:
From what I have heard OF is dying as of right now the new maintainer hasn't done any updates try the beta branch if you want to stick with OF

orangeworx, 2023-05-14:
I had read @NotExpectedYet was no longer capable of maintaining OF and had sorta handed down the reins. I dropped in a line to see if I could help myself.

Who's maintaining the beta branch? Also, do you know of any alternative to OF?
Thanks for answering

Dumnersm580, 2023-05-14:
Hello, as for a alternative yes i and a main developer of FDM-Monster it is currently a work in progress but the base system is up and running and we are working on it everyday: https://github.com/fdm-monster/fdm-monster