OctoFarm/OctoFarm #1400: OctoFarm Error ERROR (400):

3Dsoba, 2025-02-21:
### OctoFarm Version

Latest 1.2

### OctoPrint Version(s)

1,10,3

### pm2 or docker?

pm2

### Operating System

Windows

### Description of the problem

Im trying to add octoprint instance to octofarm - i enabled CORS (I REBOOTED OCTOPRINT AFTER ENABLE) and i generated manual API key
but when i try to save thiss message appears

The apikey can not be greater than 32. (but how can i control APIKEY lenght?!)

OctoFarm Error ERROR (400):
Please report this error to OctoFarm Issues!
Bad request:
###ValidationException###

#APIKEY#
The apikey can not be greater than 32.
FILE INFO
LINE: 2
COL: 665150
FILE:
http://192.168.0.81:4000/assets/js/printer-management.runner.1.7.3.min.js
STACK:
g@http://192.168.0.81:4000/assets/js/printer-management.runner.1.7.3.min.js:2:665150 @http://192.168.0.81:4000/assets/js/printer-management.runner.1.7.3.min.js:2:667626 

### Steps taken to reproduce

1. Add printer
2. Add IP adress
3. Add manualy generated API

Jarvis212121, 2025-02-23:
I have the same problem, the API key is still longer than 32 characters and it keeps giving me the error, what can I do? Help me, I'm stuck

3Dsoba, 2025-02-24:
> I have the same problem, the API key is still longer than 32 characters and it keeps giving me the error, what can I do? Help me, I'm stuck

I found the solution – first, generate the API key.
Then, close the instance (both in the browser and in CMD).
Next, go to the instance folder (mine is located at C:\Users\Administrator\AppData\Roaming\OctoPrintD1\data\), open the keys.yaml file in Notepad++, and delete the excess characters beyond 32 in the key.
Save the file, then add the newly shortened API key in OctoFarm Manager, and that's it!

closed 2025-02-24 by 3Dsoba