OctoFarm/OctoFarm #1401: octofarm.net redirects me to garbage

dschinkel, 2025-04-18:
### OctoFarm Version

latest

### OctoPrint Version(s)

latest

### pm2 or docker?

pm2

### Operating System

osx

### Description of the problem

Why is the url posted redirect me to garbage advertisements?

[octofarm.net](https://octofarm.net/)

### Steps taken to reproduce

[octofarm.net](https://octofarm.net/) is redirecting to garbage sites, what's going on here?

dschinkel, 2025-04-18:
why hasn't this been resolved?  It's been outstanding since December

https://github.com/OctoFarm/OctoFarm/issues/1399

T3chGuy77, 2025-04-30:
It seems the developer or developers have left this project. It has not been updated in over 3 years now and the domain cert was allowed to expire so the bad guys swept in and stole it to redirect to a bad website. It's proabably time to find a new Octoprint Farm manager as sooner or later this will have a security breach. Too bad the just let it go.